tianocore/edk2-test #221: Feature request for location of tests for vendor/edk2 specific protocols (Bugzilla Bug 1839)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 1839
Date: 2019-05-23T12:54:53+00:00
From: Jeff Brasen &lt;<jbrasen>&gt;
To:   Edhay &lt;<edhaya.chandran>&gt;
CC:   ashishsingha, edhaya.chandran, eric.jin, gaojie, qiaonan.liu, @cetola, supreeth.venkatesh, yonghong.zhu

Last updated: 2022-05-05T10:19:33+00:00

tianocore-issues, 2024-12-20:
## Comment 7767
Date: 2019-05-23 12:54:53 +0000
From: Jeff Brasen &lt;<jbrasen>&gt;

- Industry Specification: ---
- Release Observed: edk2-stable202105
- Releases to Fix: edk2-stable202105
- Target OS: ---
- Bugzilla Assignee(s): Edhay &lt;<edhaya.chandran>&gt;

Now that we have the edk2-test repo with the SCT infrastructure, has there been any thought for where tests that use this for non-standardized protocols should live? My initial thought is along with the protocols in edk2/edk2-platforms. For example if you had Silicon/NVIDIA/Include/Procotol/x.h you would have Silicon/NVIDIA/TestCase/x/ for this.

These could then be built and packaged in with the installed tests in the sct script. Not sure if it best to have something like IHV for these that pulls all the non-standard tests in (and then ones not for your target will be skipped as the protocol guid for that won&apos;t be present) or if vendors should construct a package with everything they are using.

tianocore-issues, 2024-12-20:
## Comment 7875
Date: 2019-05-30 20:13:50 +0000
From: Yonghong Zhu &lt;<yonghong.zhu>&gt;

Jeff Brasen, could you help to send RFC to community for discussion ?

tianocore-issues, 2024-12-20:
## Comment 8115
Date: 2019-06-19 12:17:55 +0000
From: @cetola

Bug Triage: assign to me. I&apos;ll bring this up in the next design meeting.

tianocore-issues, 2024-12-20:
## Comment 8259
Date: 2019-06-27 20:43:51 +0000
From: Liu Qiaonan &lt;<qiaonan.liu>&gt;

Stephano, please work on it.

tianocore-issues, 2024-12-20:
## Comment 8982
Date: 2019-08-13 15:28:18 +0000
From: Ashish Singhal &lt;<ashishsingha>&gt;

Stephano, Was this taken up in a design meeting?

tianocore-issues, 2024-12-20:
## Comment 9572
Date: 2019-09-10 10:27:52 +0000
From: supreeth.venkatesh

At UTWG, we agreed with the following general guidelines for doing this.

1.If the tests are for non-standardized protocols but scheduled for standardization (inside UEFI Forum based specifications) in the future, in order to be compatible with the “code first” approach, we will help create edk2-test repo inside edk2-staging repo, 
Which can be used for staging.
2.If the tests are based on standardized protocols but for a particular hardware device, we have no objections in accepting contributions that would go in IHV folder.
3.If the tests are for non-standardized protocols, it is better to host it elsewhere outside of the edk2-test as edk2-test repo is for tests against UEFI specification.

Based on this, I guess we need details from Jeff (Nvidia), 
1.What sort of tests does Nvidia have in mind for contributing?
2.Would the tests fall in any of the three categories mentioned above?

tianocore-issues, 2024-12-20:
## Comment 18838
Date: 2022-04-19 10:49:43 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

Outcome of discussion between Supreeth and Edhay:

We will document the guidelines as pescribed earlier in the edk2-test README:
1.If the tests are for non-standardized protocols but scheduled for standardization (inside UEFI Forum based specifications) in the future, in order to be compatible with the “code first” approach, we will help create edk2-test repo inside edk2-staging repo, 
Which can be used for staging.
2.If the tests are based on standardized protocols but for a particular hardware device, we have no objections in accepting contributions that would go in IHV folder.
3.If the tests are for non-standardized protocols, it is better to host it elsewhere outside of the edk2-test as edk2-test repo is for tests against UEFI specification.

Specifically for (3) &quot;non-standardized&quot; architectural protocols, the build and run of UEFI-SCT shall me controlled through a compile-time and runtime flags respectively.

And the ticket shall be closed, after a discussion in the next edk2-test triage meeting and the README update.

tianocore-issues, 2024-12-20:
## Comment 17458
Date: 2021-10-07 10:53:26 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

Edhay to consult with Supreeth and close

tianocore-issues, 2024-12-20:
## Comment 18934
Date: 2022-05-05 10:19:33 +0000
From: Barton Gao &lt;<gaojie>&gt;

HI Edhay,
Will you update the README file ?

edhay, 2025-04-03:
@edhay to check and reconfirm.
If necessary, update the README.