tianocore/edk2-test #222: Verify that loading images with <4k section alignment either succeeds or fails gracefully (Bugzilla Bug 1860)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 1860
Date: 2019-05-29T06:43:55+00:00
From: @leiflindholm
To:   Edhay &lt;<edhaya.chandran>&gt;
CC:   arvin.chen, benjamin.you, edhaya.chandran, eric.jin, gaojie, @leiflindholm, samer.el-haj-mahmoud, siyuan.fu, xypron.glpk

Last updated: 2022-04-19T10:51:08+00:00

tianocore-issues, 2024-12-20:
## Comment 7851
Date: 2019-05-29 06:43:55 +0000
From: @leiflindholm

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: Other
- Bugzilla Assignee(s): Edhay &lt;<edhaya.chandran>&gt;

We have seen issues in the field where platforms that enable NX attribute on the pe/coff header hang hard when presented.

So it would be beneficial if we had a test that verified graceful failure (or success) of images  with &lt;4k section alignment.

tianocore-issues, 2024-12-20:
## Comment 7916
Date: 2019-06-03 14:50:12 +0000
From: supreeth.venkatesh

Acknowledged. 
Queued for development.

tianocore-issues, 2024-12-20:
## Comment 9818
Date: 2019-09-25 19:04:53 +0000
From: supreeth.venkatesh

Can you please point me to relevant section in the UEFI Specification which mandates section alignment restrictions so that additional test case can be added to existing image tests?

tianocore-issues, 2024-12-20:
## Comment 9822
Date: 2019-09-25 20:09:28 +0000
From: @leiflindholm

There is no explicit statement about the alignment.

There is an *implicit* requirement that if you want to use EFI_MEMORY_XP attribute, you need to keep non-executable regions in separate pages from executable regions - for example ensuring the page containing the PE header does not also contain code.

The request is for a test that ensures that *if* a platform enforces restrictions on alignment, then it fails gracefully when an image with smaller alignment is executed.

Today we have platforms in the wild that hang hard if you execute a program with &lt; 4k section alignment.

tianocore-issues, 2024-12-20:
## Comment 16231
Date: 2021-06-01 07:54:34 +0000
From: @leiflindholm

Supreeth left ARM a while ago - re-assigning to Edhay for consideration.

tianocore-issues, 2024-12-20:
## Comment 15869
Date: 2021-05-01 04:40:51 +0000
From: Heinrich Schuchardt &lt;<xypron.glpk>&gt;

@Leif:
Will you provide a patch?

tianocore-issues, 2024-12-20:
## Comment 18839
Date: 2022-04-19 10:51:08 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

@Lief: Can your team provide a patch for this solution?

shkrc, 2026-04-10:
Hi, let me add some of my observations versus whats been discussed in this ticket

1) I had set the PcdImageProtectionPolicy to be 0x3 in my codebase, where the bits represent the following
     BIT0       - Image from unknown device. <BR>
     BIT1       - Image from firmware volume.<BR>
Now, I tried compiling SCT image, put it in a USB drive and tried to boot it and was getting this error

```
ProtectUefiImageCommon - 0x6AB72D40
  - 0x000000006A5CA000 - 0x0000000000112B00
!!!!!!!!  Image Section Alignment(0x40) does not match Required Alignment (0x1000)  !!!!!!!!
ProtectUefiImage failed to create image properties record
```

The reason was that EDK2's image protection policy checks for 4k alignment in PE headers of efi binaries, and the SCT binary didnt have its PE header 4k aligned.

Possible solution: To make the SCT binary PE headers align with 4k.

2) This ticket talks about writing new test case that checks BIOS if the 4k enforcement results in a non-aligned image gracefully failing rather than hang.

In our case, the SCT binary itself fails the alignment. I would say the requirement 1 and 2 are related, and once 1 is fixed, we can discuss on how to proceed having a solution for case 2.