tianocore/edk2-test #224: NULL dereference in BBTestQueryFunctionAutoTest() (Bugzilla Bug 2238)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 2238
Date: 2019-10-03T04:14:42+00:00
From: Heinrich Schuchardt &lt;<xypron.glpk>&gt;
To:   Edhay &lt;<edhaya.chandran>&gt;
CC:   edhaya.chandran, eric.jin, nobody, supreeth.venkatesh, xypron.glpk

Last updated: 2022-01-13T08:17:16+00:00

tianocore-issues, 2024-12-20:
## Comment 9915
Date: 2019-10-03 04:14:42 +0000
From: Heinrich Schuchardt &lt;<xypron.glpk>&gt;

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): Edhay &lt;<edhaya.chandran>&gt;

Function BBTestQueryFunctionAutoTest() has:

EFI_GUID *ParameterTypeGuid=NULL;
*ParameterTypeGuid =gBlackBoxEfiPlatformToDriverConfigurationClpGuid;

So here we are writing to address 0x0.

In the same function

UINTN                                          *ParameterBlockSize=0;
*ParameterBlockSize = sizeof( EFI_CONFIGURE_CLP_PARAMETER_BLK );

The errors were indicated by CppCheck (http://cppcheck.sourceforge.net/). There are more errors of the the same kind in the code. Please, run CppCheck on the whole of edk2-test and fix them all.

Best regards

Heinrich Schuchardt

tianocore-issues, 2024-12-20:
## Comment 10147
Date: 2019-10-21 12:52:08 +0000
From: supreeth.venkatesh

In pipeline for next release.

tianocore-issues, 2024-12-20:
## Comment 13455
Date: 2020-08-18 22:23:31 +0000
From: nobody &lt;<nobody>&gt;

Edhay, please take it.

tianocore-issues, 2024-12-20:
## Comment 17181
Date: 2021-09-07 04:56:25 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

Vincent provide the initial static check reports of such issues. Later the actual fixes shall be distributed among the community volunteers.

tianocore-issues, 2024-12-20:
## Comment 17149
Date: 2021-09-02 10:50:54 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

*** Bug 2237 has been marked as a duplicate of this bug. ***

tianocore-issues, 2024-12-20:
## Comment 18123
Date: 2022-01-13 08:17:16 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

Vincent has provided the report, they are uploaded in the below ticket
https://bugzilla.tianocore.org/show_bug.cgi?id=2239

edhay, 2025-04-03:
The Coverity report has been provided by Vincent and the fixes are tracked here:
https://github.com/tianocore/edk2-test/issues/233

Action: @edhay to fix this particular code issue and then close the ticket.