tianocore/edk2-test #225: UpdateInfoFileName() returns random bytes from stack (Bugzilla Bug 2239)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 2239
Date: 2019-10-03T04:24:09+00:00
From: Heinrich Schuchardt &lt;<xypron.glpk>&gt;
To:   Edhay &lt;<edhaya.chandran>&gt;
CC:   edhaya.chandran, eric.jin, @cetola, supreeth.venkatesh, xypron.glpk

Last updated: 2021-11-11T09:47:04+00:00

tianocore-issues, 2024-12-20:
## Comment 9916
Date: 2019-10-03 04:24:09 +0000
From: Heinrich Schuchardt &lt;<xypron.glpk>&gt;

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): Edhay &lt;<edhaya.chandran>&gt;

In some cases (Size &lt;= FileInfo-&gt;Size) UpdateInfoFileName() does not initialize the variable status. So random bytes from the stack will be returned.

I guess that status should be initalized to EFI_SUCCESS.

The problem was indicated by CppCheck (http://cppcheck.sourceforge.net/).

Please, run cppcheck on the whole of edk2-test to discover a lot of similar problems.

tianocore-issues, 2024-12-20:
## Comment 10148
Date: 2019-10-21 12:52:26 +0000
From: supreeth.venkatesh

In pipeline for next release.

tianocore-issues, 2024-12-20:
## Comment 17147
Date: 2021-09-02 10:49:14 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

Vincent to run cppcheck/Coverity and prepare the list of issue

tianocore-issues, 2024-12-20:
## Comment 17175
Date: 2021-09-06 09:04:32 +0000
From: Vincent Stehlé &lt;<vincent.stehle>&gt;

Created attachment 822
Cppcheck results

Cppcheck 2.3
edk2-test 6756aa257d1b418e19dd21031e613a8970ffada6

Attachment: [BZ-822-cppcheck.log](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2-test/BZ-822-cppcheck.log)

tianocore-issues, 2024-12-20:
## Comment 17452
Date: 2021-10-07 10:21:22 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

Carolyn and Edhay to Volunteer to fix the CPPCheck Issues found.

edhay, 2025-04-03:
The community has take up fixing the Coverity report generated on edk2-test code.
This is tracked by https://github.com/tianocore/edk2-test/issues/233
This ticket is a duplicate and hence closing.

edhay, 2025-04-03:
@edhay Solve the UpdateInfoFileName() issue.