tianocore/edk2-test #226: SctPkg does not build for a variety of architectures/toolchains (Bugzilla Bug 3595)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 3595
Date: 2021-08-26T17:34:03+00:00
From: @heatd
To:   Carolyn Gjertsen &lt;<Carolyn.Gjertsen>&gt;
CC:   edhaya.chandran, gaojie, @heatd

Last updated: 2022-05-05T10:04:21+00:00

tianocore-issues, 2024-12-20:
## Comment 17092
Date: 2021-08-26 17:34:03 +0000
From: @heatd

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): Carolyn Gjertsen &lt;<Carolyn.Gjertsen>&gt;

Due to a variety of issues throughout SctPkg, multiple architecture/toolchain combos are unable to build UEFI SCT.

Such combinations include:
- GCC5  (IA32 and X64)
- GCC49 (IA32 and X64)
- VS2019(IA32 and X64)
- CLANGPDB(IA32 and X64)

I managed to patch SctPkg so it builds for VS2019(both IA32 and X64) and
GCC5(IA32 only) and I&apos;m keeping the patched version at my edk2-test fork:
https://github.com/heatd/edk2-test; I might submit these patches to the mailing list soon.

Unfortunately, for GCC5 X64 there are a lot of EFIAPI issues that stop it from building (see Bug 2126 - SCT does not build for X64 using GCC - missing EFIAPI); I tried my hand at fixing those but there are way too many issues/warnings, and I&apos;m not too familiar with this project.

CLANGPDB fails to build SctPkg in the IA32 and X64 architectures.

If more details are required I am more than happy to submit further bug reports.

Relevant toolchain information:

- Visual Studio Community 2019
- GCC version 11.1.0 (GCC)
- GNU ld (GNU Binutils) 2.36.1 / GNU gold (GNU Binutils 2.36.1) 1.16
- Clang version 12.0.1 / LLD 12.0.1

tianocore-issues, 2024-12-20:
## Comment 17104
Date: 2021-08-27 16:56:39 +0000
From: @heatd

CC&apos;d SctPkg maintainers

tianocore-issues, 2024-12-20:
## Comment 17138
Date: 2021-09-02 10:26:52 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

Carolyn to further check

tianocore-issues, 2024-12-20:
## Comment 18932
Date: 2022-05-05 10:04:21 +0000
From: Carolyn Gjertsen &lt;<Carolyn.Gjertsen>&gt;

Still in progress.  Will push changes to begin feedback process.

tianocore-issues, 2024-12-20:
## Comment 17140
Date: 2021-09-02 10:29:49 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

Confirmed

edhay, 2025-04-03:
@cgjertsen to check and re-confirm.

sunnywang-arm, 2025-05-14:
It looks like X64 build issue with GCC will be solved by https://github.com/tianocore/edk2-test/issues/244 
Cc @MarsDoge

sunnywang-arm, 2025-05-14:
Do we still need to maintain the build support for VS20xx and CLANGPDB? If not, I propose dropping them to save our maintenance effort. 
- For VS20xx,  Linux OS (e.g., Ubuntu) can now be run on Windows OS through WSL (Windows Subsystem for Linux), which means users can use Linux+GCC in Windows OS to build the SCT, so I don't see a need to continue maintaining VS20xx. 
- As for CLANGPDB, it is not mentioned in our build instructions (HowToBuildSctInUdk2017.txt and HowToBuildSct.txt), so it's unlikely that anyone is relying on it.

In short, I think only having Linux+GCC should be good enough. 

Cc @edhay @cgjertsen

vishalo, 2025-06-04:
@sunnywang-arm Linux + CLANG (DWARF) is also useful.  I will submit a patch for this.
Cc @edhay

vishalo, 2025-06-05:
Thank you @sunnywang-arm - glad we agree on this. Yes, we do use CLANGDWARF for edk2-test, and we would need to address build failures - currently there are none. I think SCT should support all toolchains that edk2 supports, otherwise it is a hassle to support different compilers for different packages.