tianocore/edk2-test #231: UpdateCapsule Conformance SCT does not use a known good capsule header guid (Bugzilla Bug 4098)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4098
Date: 2022-10-05T19:24:28+00:00
From: @javagedes
To:   @javagedes
CC:   edhaya.chandran

Last updated: 2024-09-05T10:37:56+00:00

tianocore-issues, 2024-12-20:
## Comment 19718
Date: 2022-10-05 19:24:28 +0000
From: @javagedes

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @javagedes

UpdateCapsule() Conformance requires that a capsule marked CAPSULE_FLAGS_PERSIST_ACROSS_RESET must also provide a valid ScatterGatherList. In some platforms (See example [https://github.com/microsoft/mu_basecore/blob/2f7175d6bb584d5de26d6d7deb0b6da5ff6dd5c0/MdeModulePkg/Universal/CapsuleRuntimeDxe/CapsuleService.c#L182]), however, if a capsule is marked as such, the capsule is validated before being persisted in memory (which is when the ScatterGatherList is verified). Due to this, those platforms fail the SCT that validates this conformance as they return EFI_UNSUPPORTED as the capsule header guid being passed to the function is not a known valid capsule header guid.

This commit updates the capsule header guid used in the MiscRuntimeService tests to use a known good capsule header guid, gWindowsUxCapsuleGuid, which causes these platforms to pass this SCT by reaching the ScatterGatherList validation, which then returns EFI_INVALID_PARAMETER.

tianocore-issues, 2024-12-20:
## Comment 19719
Date: 2022-10-05 19:27:29 +0000
From: @javagedes

Created attachment 1034
Patch

Attachment: [BZ-1034-patch.patch](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2-test/BZ-1034-patch.patch)

tianocore-issues, 2024-12-20:
## Comment 19727
Date: 2022-10-06 11:36:57 +0000
From: @javagedes

Yes. The simple description of the edge case is that if a capsule has the flag CAPSULE_FLAGS_PERSIST_ACROSS_RESET set, we immediately verify we support the capsule by executing the function StageCapsuleImage() which in turn calls SupportCapsuleImage(). 

The capsule passed to the function fails. There are multiple ways to pass this check (See Edk2&apos;s SupportCapuleImage() https://github.com/tianocore/edk2/blob/710f83b79d6eab641401c054b2f40f6c630f8cd5/MdeModulePkg/Library/DxeCapsuleLibFmp/DxeCapsuleLib.c#L1499), but the easiest in my opinion is to use the gWindowsUxCapsuleGuid. 

By making the change in the patch, the guid passes this check and makes it to the function that validates the ScatterGatherList, where it is expected to fail with an EFI_INVALID_PARAMETER.

Feel free to reach out if you have any questions.

tianocore-issues, 2024-12-20:
## Comment 21620
Date: 2023-06-01 10:08:31 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

@Joey Vagedes : Do you have any updates on the previous comment.

tianocore-issues, 2024-12-20:
## Comment 19927
Date: 2022-11-03 10:48:34 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

The behavior for Capsule Update w.r.t to Capsule Header GUID to be checked offline

SupportCapsuleImage()
{
}

StageImage()
{
   if(!SupportCapsuleImage())
   {
      return EFI_UNSUPPORTED;
   }
}

1. Can we query the SystemTable for the GUID, and set the value for the test parameter
2. Is there a SCT test that checks the validity of the Capsule?

tianocore-issues, 2024-12-20:
## Comment 23382
Date: 2024-09-05 10:37:56 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

Edhay to reach to Microsoft for any updates on their solution.

edhay, 2025-04-03:
@edhay to follow up offline with Microsoft engineers.

sunnywang-arm, 2025-07-03:
I think the solution from Microsoft (https://github.com/tianocore/edk2-test/pull/66) may cause test failure with the capsule update firmware implementation that is based on other valid Capsule header GUIDs.  

Therefore, another solution is to update the test to cover four valid Capsule header GUIDs listed below. The test can be updated to call gtRT->UpdateCapsule() with these GUIDs one by one, and report a failure if it doesn't get the expected result from any of the four calls. 
- EFI_CAPSULE_GUID - {3B6686BD-0D76-4030-B70E-B5519E2FC5A0}  
- WINDOWS_UX_CAPSULE_GUID - {3B8C8162-188C-46A4-AEC9-BE43F1D65697}
- EFI_FIRMWARE_MANAGEMENT_CAPSULE_ID_GUID - {6DCBD5ED-E82D-4C44-BDA1-7194199AD92A}  
- EFI_JSON_CAPSULE_ID_GUID - {0x67d6f4cd, 0xd6b8,  0x4573, {0xbf, 0x4a, 0xde, 0x5e, 0x25, 0x2d, 0x61, 0xae }}

Cc @edhay