tianocore/edk2-test #233: Generate Coverity static code check report on edk2-test and fix (Bugzilla Bug 4844)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4844
Date: 2024-09-06T16:35:09+00:00
From: Edhay &lt;<edhaya.chandran>&gt;
To:   unassigned &lt;<unassigned>&gt;
CC:   vincent.stehle

Last updated: 2024-11-07T10:45:52+00:00

tianocore-issues, 2024-12-20:
## Comment 23388
Date: 2024-09-06 16:35:09 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): unassigned &lt;<unassigned>&gt;

Generate the Coverity static code check report on edk2-test code and fix high impact and high return errors/warnings.

tianocore-issues, 2024-12-20:
## Comment 23424
Date: 2024-10-03 10:36:11 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

Vincent to provide a Coverity run report on the latest edk2-test code.

To plan: Once the issues have been resolved, the UEFI-SCT must be tested for any impact due to the fixes.

tianocore-issues, 2024-12-20:
## Comment 23429
Date: 2024-10-04 09:24:56 +0000
From: Vincent Stehlé &lt;<vincent.stehle>&gt;

Created attachment 1543
Coverity analysis results in oneline text format

Attachment: [BZ-1543-oneline.zip](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2-test/BZ-1543-oneline.zip)

tianocore-issues, 2024-12-20:
## Comment 23428
Date: 2024-10-04 09:23:58 +0000
From: Vincent Stehlé &lt;<vincent.stehle>&gt;

Created attachment 1542
Coverity analysis summary

Attachment: [BZ-1542-summary.txt](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2-test/BZ-1542-summary.txt)

tianocore-issues, 2024-12-20:
## Comment 23430
Date: 2024-10-04 09:25:52 +0000
From: Vincent Stehlé &lt;<vincent.stehle>&gt;

Created attachment 1544
Coverity analysis results in JSON format

Attachment: [BZ-1544-json.zip](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2-test/BZ-1544-json.zip)

tianocore-issues, 2024-12-20:
## Comment 23431
Date: 2024-10-04 09:26:55 +0000
From: Vincent Stehlé &lt;<vincent.stehle>&gt;

Created attachment 1545
Coverity analysis results in MSVS format

Attachment: [BZ-1545-msvs.zip](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2-test/BZ-1545-msvs.zip)

tianocore-issues, 2024-12-20:
## Comment 23432
Date: 2024-10-04 09:32:14 +0000
From: Vincent Stehlé &lt;<vincent.stehle>&gt;

Hi,

I have built the SCT for X86 and AARCH64 under Coverity. The results in various formats are attached to this issue.

Due to the attachment size limit of 1 MB, it is sadly not practicable to attach the results in HTML format here, as they weight ~80 MB.

This is with:
- Coverity 2023.9.0
- edk2 91d806917fd2
- edk2-test 16c552d70b74

tianocore-issues, 2024-12-20:
## Comment 23519
Date: 2024-11-07 10:45:52 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

Thank you, Vincent for the reports.

Ann to take up the initial analysis of the reports.
The fix effort shall be shared among community members.

edhay, 2025-04-03:
Update: The untinitialised variables errors are resolved and draft reviewed.
To be upstreamed by @edhay in the mailing list for community review.