tianocore/edk2-test #234: uefi-sct: Incorrect test validation of EFI_SIMPLE_TEXT_INPUT_EX_PROTOCOL.SetState() in ReadKeyStrokeEx Conformance Test (Bugzilla Bug 4862)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4862
Date: 2024-10-08T14:02:24+00:00
From: Edhay &lt;<edhaya.chandran>&gt;
To:   Edhay &lt;<edhaya.chandran>&gt;
CC:   edhaya.chandran, Sunny.Wang, vincent.stehle

Last updated: 2024-11-07T16:47:44+00:00

tianocore-issues, 2024-12-20:
## Comment 23518
Date: 2024-11-07 10:38:14 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

The issue is valid.
Assigned to Edhay.

tianocore-issues, 2024-12-20:
## Comment 23434
Date: 2024-10-08 14:02:24 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): Edhay &lt;<edhaya.chandran>&gt;

UEFI specification does not require the setting of toggle states to be supported. This holds true even if EFI_KEY_STATE_EXPOSED is supported.

The spec does not describe what passing or not passing EFI_TOGGLE_STATE_VALID to SetState() will result in. This should be clarified in the specification.

EFI_SIMPLE_TEXT_INPUT_EX_PROTOCOL.SetState() returning EFI_UNSUPPORTED must be accepted as a valid return value. It does not constitute a bug.

The solution in https://bugzilla.tianocore.org/show_bug.cgi?id=3390
must be extended to  ReadKeyStrokeEx Conformance Test

https://github.com/tianocore/edk2-test/blob/master/uefi-sct/SctPkg/TestCase/UEFI/EFI/Protocol/SimpleTextInputEx/BlackBoxTest/SimpleTextInputExBBTestConformance.c#L494

Presently this test prints a warning when toggle state is passed as NULL.

tianocore-issues, 2024-12-20:
## Comment 23521
Date: 2024-11-07 11:20:26 +0000
From: Sunny Wang &lt;<Sunny.Wang>&gt;

Took a further look into this one after our meeting. I think the problem is that the whole test BBTestSetStateConformanceTestCheckpoint1 is invalid or wrong. Just like you mentioned in the comment, the spec does not describe what passing or not passing EFI_TOGGLE_STATE_VALID to SetState() will result in.  Actually the UEFI spec doesn&apos;t mention EFI_INVALID_PARAMETER as one of SetState() returned status codes, so I think this test should either be removed or be changed to check EFI_UNSUPPORTED instead.

tianocore-issues, 2024-12-20:
## Comment 23526
Date: 2024-11-07 16:46:44 +0000
From: Edhay &lt;<edhaya.chandran>&gt;

Thank you, Sunny for the update.
Yes, I agree that the test itself may need to change.  

Presently the spec supports the below return values for SetState():
EFI_SUCCESS     : The device state was set appropriately.
EFI_DEVICE_ERROR: The device is not functioning correctly and could not have the setting adjusted.
EFI_UNSUPPORTED : The device does not support the ability to have its state set or the requested state change was not supported.

For the NULL input,
my proposal is to change the test to below:
If return value EFI_SUCCESS     : Print error
If return value EFI_DEVICE_ERROR: Print warning. Device itself has an error. Ideally even for non-NULL values the device should return the same if its not functioning correctly. 
If return value EFI_UNSUPPORTED : Print Success

Please let me know your thoughts.

edhay, 2025-04-03:
@sunnywang-arm to check the analysis offline and give his suggestion.

sunnywang-arm, 2025-04-25:
@edhay  Sorry for the delay. The solution you proposed looks good to me.