tianocore/edk2-test #240: uefi-sct/SctPkg: EFI_ADAPTER_INFORMATION_PROTOCOL Test

Archi1986, 2025-01-24:
The EFI_ADAPTER_INFORMATION_PROTOCOL Test 5.5.11.1.2 always FAILS with the AIP Support Type gEfiAdapterInfoNetworkBootGuid.

In the Guid C file, gEfiAdapterInfoUndiIPv6SupportGuid is assigned to an incorrect Guid.

Thanks.

uefi-sct\SctPkg\TestCase\UEFI\EFI\Protocol\AdapterInfo\BlackBoxTest\AdapterInfoBBTestMain.c
```
BOOLEAN
InfoTypeAndBlockSizeCmp( 
  EFI_GUID    InformationType,
  UINTN       InformationBlockSize
  )
{
   
  if ((!GuidCmp (InformationType,gEfiAdapterInfoMediaStateGuid) && InformationBlockSize != sizeof(EFI_ADAPTER_INFO_MEDIA_STATE)) || 
      (!GuidCmp (InformationType,gEfiAdapterInfoNetworkBootGuid) && InformationBlockSize != sizeof(EFI_ADAPTER_INFO_NETWORK_BOOT)) || 
      (!GuidCmp (InformationType,gEfiAdapterInfoSanMacAddressGuid) && InformationBlockSize != sizeof(EFI_ADAPTER_INFO_SAN_MAC_ADDRESS)) ||
      (!GuidCmp (InformationType,gEfiAdapterInfoUndiIPv6SupportGuid) && InformationBlockSize != sizeof(EFI_ADAPTER_INFO_UNDI_IPV6_SUPPORT)) ||
      (!GuidCmp (InformationType,gEfiAdapterInfoMediaTypeGuid) && InformationBlockSize != sizeof(EFI_ADAPTER_INFO_MEDIA_TYPE))) 
    return FALSE;
  else
    return TRUE;  
}
```

uefi-sct\SctPkg\TestCase\UEFI\EFI\Protocol\AdapterInfo\BlackBoxTest\Guid.c
```
EFI_GUID gEfiAdapterInfoMediaStateGuid              = EFI_ADAPTER_INFO_MEDIA_STATE_GUID;

EFI_GUID gEfiAdapterInfoSanMacAddressGuid           = EFI_ADAPTER_INFO_SAN_MAC_ADDRESS_GUID;

EFI_GUID gEfiAdapterInfoNetworkBootGuid             = EFI_ADAPTER_INFO_NETWORK_BOOT_GUID;

EFI_GUID gEfiAdapterInfoUndiIPv6SupportGuid         = EFI_ADAPTER_INFO_NETWORK_BOOT_GUID;
```

edhay, 2025-02-06:
Hi @Archi1986 , Thank you for reporting this bug.
Could you please submit the solution patch for the same.

Archi1986, 2025-07-04:
I don't have any further questions.

Archi1986, 2025-07-05:
The issue in the EFI_ADAPTER_INFORMATION_PROTOCOL Test 5.5.11.1.2 has been resolved.
If there are any other topics that need to be discussed, please feel free to open a new thread.
Thank you all for your support!

closed 2025-07-05 by Archi1986