tianocore/edk2-test #250: [Feature]: Review and add new warning flags in UEFI-SCT build

edhay, 2025-04-28:
### Feature Overview

Review the existing warning flags in UEFI-SCT compiling. Add more capture more potential errors.

For e.g. The issues such as 
https://github.com/tianocore/edk2-test/pull/247

Should be flagged early. 

### Solution Overview

 Stricter warning flags (-Werror, -Wimplicit-function-declaration, etc.) in the SCT build could be added, so we don't see a fatal error when a header file is missing.

### What are impacted?

Build scripts

### Urgency

Medium

### Are you going to implement the feature request?

Someone else needs to implement the feature

### Do you need maintainer feedback?

Maintainer feedback requested

sunnywang-arm, 2025-07-02:
https://github.com/tianocore/edk2-test/pull/100

edhay, 2025-10-02:
Looking for Volunteers to fix