tianocore/edk2-test #252: [Feature]: [New test coverage] Add two new tests for GetImageInfo() in FirmwareManagementBBTestConformance

edhay, 2025-05-13:
### Feature Overview

The clarifications added in UEFI Spec via the ECR: https://mantis.uefi.org/mantis/view.php?id=1869 
warrants two new tests to cover these cases for EFI_FIRMWARE_MANAGEMENT_PROTOCOL.GetImageInfo()

Case 1: ImageInfoSize = 0 and ImageSize = NULL (This is a valid case)
            Expected return: EFI_BUFFER_TOO_SMALL and ImageInfoSize > 1

Case 2: ImageInfoSize = 0 and ImageSize = <not too small> (This is a invalid case)
             Expected return: EFI_INVALID_PARAMETER

### Solution Overview

Two new tests to be added in edk2-test\uefi-sct\SctPkg\TestCase\UEFI\EFI\Protocol\FirmwareManagement\BlackBoxTest\FirmwareManagementBBTestConformance.c to cover the above cases.

### Alternatives Considered

NA

### What are impacted?

Test Coverage

### Urgency

Medium

### Are you going to implement the feature request?

I will implement the feature

### Do you need maintainer feedback?

Maintainer feedback requested

### Anything else?

The documentation in SCT test case spec should be updated with the two new tests

edhay, 2025-05-13:
The new scenarios are drafted for review.
https://github.com/tianocore/edk2-test/pull/254

The code shall be upstreamed once the scenarios are reviewed.

sunnywang-arm, 2025-05-14:
Hi @edhay 
Now it seems we only have a pull request for the document change. I can't find a code change for this. Should we keep this issue open until the code gets upstreamed?

edhay, 2025-05-14:
Hi @sunnywang-arm ,  Yes, only the documented scenarios was upstreamed.
Thank you for reviewing the same.

Will upstream the code soon.
Will keep this issue open till then.

edhay, 2025-05-20:
The code for the two scenarios are as in the PR: 
https://github.com/tianocore/edk2-test/pull/255

Verification log is as attached. 

[Conformance_Test-GetImageInfo_0_0_54EE347C-10A3-4311-9427-A897BFDC9808.log](https://github.com/user-attachments/files/20321082/Conformance_Test-GetImageInfo_0_0_54EE347C-10A3-4311-9427-A897BFDC9808.log)

The new tests logs are:

> GetImageInfo, conformance checkpoint #3 -- PASS
> 2C6AA8E6-3356-43F4-BCCA-F7DC4D20BBF3
> Result - correctly returned EFI_BUFFER_TOO_SMALL.
> /home/edhcha01/EDHAY-FORK/SctPkg/TestCase/UEFI/EFI/Protocol/FirmwareManagement/BlackBoxTest/FirmwareManagementBBTestConformance.c:1094: Status - Buffer Too Small
> 
> GetImageInfo, conformance checkpoint #4 -- PASS
> 8EE98EB3-EE5D-4182-A297-2A0082C5E952
> Result - correctly returned EFI_INVALID_PARAMETER.
> /home/edhcha01/EDHAY-FORK/SctPkg/TestCase/UEFI/EFI/Protocol/FirmwareManagement/BlackBoxTest/FirmwareManagementBBTestConformance.c:1191: Status - Invalid Parameter

edhay, 2025-05-21:
The solution is reviewed and upstream by the commit: 
https://github.com/tianocore/edk2-test/commit/40cc44e557fbebb0319bc30b1dfdaca7072f7260

closed 2025-05-21 by edhay